MechanicalVentilatorMilano/gui #11: add a "stop" button

carmelopellegrino, 2020-03-30:
The button should stay in a place of the screen where is hard to get clicked by mistake, for example the top-right corner of the window.

The action when clicked must be to show a popup window with a label and two buttons:
- "Cancel", if clicked close the popup and continue the program as before (big green button); 
-  "STOP", if clicked sends the stop signal to the ESP32, the program may continue running (smaller red all-CAPS button);
- the label must asking for confirmation with a clear statement of what's going to happen if the uses clicks on the "STOP" button.

javerom, 2020-03-30:
Is it sufficient to keep the current layout (i.e. Start becomes Stop and grays out other option), but add the confirmation to Stop?

carmelopellegrino, 2020-03-30:
In think that what you suggest would also work. Thinking from the ux point of view, could be possible to pass through the following?
- click on start (any of the two)
- both buttons get gray
- after a 2 seconds delay (tunable? config file? max 3 seconds), the stop button becomes visible in red (light one, not to be confused with an alarm but clearly reddish, #fc6203?) and clickable

javerom, 2020-03-30:
The above has been implemented. Awaiting pull request.

closed 2020-03-30 by javerom