MechanicalVentilatorMilano/gui #18: Add a fake ESP32 class for testing

bs610, 2020-03-30:
I don't have a physical ESP32 to test with, but a lot of GUI development doesn't require it. Adding a FakeESP32Serial class that simulates get/set will help with this.

carmelopellegrino, 2020-03-30:
do you have an arduino (any model)?

carmelopellegrino, 2020-03-30:
anyway, there is also an option that can be set in the `yaml` config file to disable the use of the serial connection. I thing that should be the attack surface.

javerom, 2020-03-30:
I had troubles with this earlier and set up a virtual COM port. However I agree that the solution is wrap all the serial stuff in the config file option

bs610, 2020-03-30:
I don't have an Arduino at all, but will hopefully have one later in the week. Currently read_from_esp=False disables the IO thread completely, which changes the program behaviour quite a lot (no data filling etc).

I have a branch where I make read_from_esp=False use a dummy class (with get/set functions), so the rest of the program behaves as normal, just with junk data.

If this sounds like an acceptable approach, can you give my account permission to push branches to github?

carmelopellegrino, 2020-03-30:
OK Ben, let me just suggest the approach.

In `gui/mvm_gui.py` there is the instantiation of the `esp32` object. Initialise this variable so that:
* if `read_from_esp=True` in the configuration file, the esp32 object is of the `ESP32Serial` type exactly like it is now;
* otherwise, it is a different class that provides `set` and `get` functions. Bear in mind that the code may at some point `set` a value and test if it is still the same value using a `get`. A `dict` would do the job.
Moreover, `get("mve")` , `get("vti")` and `get("vte")` shall return floating random values between 10 and 100.

Thank you for your support!

closed 2020-03-30 by carmelopellegrino