MechanicalVentilatorMilano/gui #41: target platform?

tsengj10, 2020-03-31:
Sorry, this was probably discussed earlier, but on the general call today, someone mentioned having training software on tablets.

Are we aiming to have the gui running on tablets, or will it always be running on a PC?

And if we should be targeting tablets, do we have a python-on-android/iOS solution in mind which will support a graphical user interface?

carmelopellegrino, 2020-03-31:
Hi Jeff,

I didn't participate to the today general meeting so I missed this news.

Training could be done in many different ways, using for example standard PCs, which are currently supported. For example, we have for sure got it run on both macOS and Linux, apart from Raspberry Pi, and I'm quite confident that eventually it will work on Windows.

So the answer is no: we don't have a python-on-android/iOS solution in mind.

tsengj10, 2020-03-31:
Thanks.  We may want to get some clarity on this at some point, i.e., what device platforms are most common where the ventilator might be used.  In the meantime, however, it shouldn't affect current development; worst case, the current code is the invaluable prototype which will guide any re-coding.

carmelopellegrino, 2020-03-31:
Thank you for having raised the point!
I close the issue for now.

closed 2020-03-31 by carmelopellegrino