MechanicalVentilatorMilano/gui #60: presets

carmelopellegrino, 2020-04-01:
- [ ] for each settings parameter, add a `presets` property as a list of values in the yaml configuration file, e.g.
```yaml
respiratory_rate:
    name: "Resp. Rate"
    default: 12
    min: 4
    max: 50
    step: 1
    current: None
    units: "b/min"
    presets:
      - 10
      - 20
      - 30
      - 35
```
- [ ] create a `Preset Selector` popup window that:
- opens up when the user clicks on a value input box in the settings window
- contains a button for each preset value in a stack or matrix and a "Cancel" button
- [ ] on click on "Cancel", just close the popup window
- [ ] on click on one of the other buttons, close the popup window and set the corresponding value in the input box

carmelopellegrino, 2020-04-01:
on paper preview :)

[image]

marcodeltutto, 2020-04-01:
I can start working on this

PR #67 (2020-04-02): Fixed #60
 - fixes #60 
 - when the user clicks on a setting input box, a panel opens showing the possible preset values
commit a349a57b (2020-04-02): Merge pull request #67 from MVMSoftware/issue-60

Fixed #60
closed 2020-04-02 by carmelopellegrino