MechanicalVentilatorMilano/gui #78: Connect pause buttons in menu

bs610, 2020-04-03:
The behaviour of pause buttons is:

- on press, set a parameter to 1 to the ESP32
- on release, set the same parameter to 0 to the ESP32

bs610, 2020-04-03:
We discussed this at a Canadian meeting, and are concerned that "0" is only sent when the button is released. If the GUI crashes while the button is pressed, then the patient is left in a very bad state.

It seems like it would be better to continually send 1 while the button is pressed, and if the ESP32 notes that it hasn't seen a 1 for a while (a few seconds?) then it returns to normal state. I guess this needs to be discussed with those writing the ESP32 code.

carmelopellegrino, 2020-04-04:
@bs610 Excellent point! This is something to be addressed to the firmware developer(s). So I think that for the moment we could keep this issue in the background and revive it when this point will be cleared.

carmelopellegrino, 2020-04-08:
Latest info about it:

- [ ] start the inspiratory pause on the Qt `press` event:
```
set pause_inhale 1
```
and **keep** setting every 100ms until a `release` event. On `release`, immediately stop sending `set pause_inhale 1` and send `set pause_inhale 0`.

- [ ] start the expiratory pause on the Qt `press` event:
```
set pause_exhale 1
```
and **keep** setting every 100ms until a `release` event. On `release`, immediately stop sending `set pause_exhale 1` and send `set pause_exhale 0`.

PR #106 (2020-04-08): Inspiration expiration buttons connected
- Fixes #78.
- The "pause inspiration" and "pause expiration" buttons are connected to the appropriate worker function.
- When pressed, a signal `set pause_inhale 1` or `set pause_exhale 1` is sent to the ESP every 100 ms (configurable via yaml param `expinsp_setinterval`). This is done via a QTimer.
- When the button is released, the QTimer stops.
- If we cannot communicate to the ESP, a critical box is raised, with the only option "Ok".
closed 2020-04-08 by carmelopellegrino