MechanicalVentilatorMilano/gui #85: in settings window, make extremely clear the mode the below parameters are set for

carmelopellegrino, 2020-04-03:
it might be hard (opinions?) to understand from the tab bar the mode the parameters below are set for.

Possible solutions (more ideas are welcome):
- change the color of the background of the tab label to improve visibility
- add an additional label on top of the parameters listing.

javerom, 2020-04-03:
I've been giving some thought to the settings window in general. For this topic another option could be to switch to a Stacked widget so that we can control the size and visibility of the buttons used per mode.

More broadly, I've been thinking about incorporating the settings window within the main window, where clicking "Settings" from the main window would just replace the plot and monitor area with the current settings layout. This way, we'd be able to keep the bottom bar, which could have information like Current Mode, Current state (running vs not), Set Auto/Assist, and Start/Stop.

The GUI may be too integrated for such a big change, but I think it could be done in a couple hours.

Thoughts?

carmelopellegrino, 2020-04-10:
- [ ] let's go with colouring the active and inactive tabs differently. It's clear enough this way and easy to implement. 
- [ ] I'll also add a permanent label below the "backup"-related widgets in assisted mode the stating that the automatic-mode settings are in the other tab.

carmelopellegrino, 2020-04-10:
here is a reference on how to colour the tabs https://doc.qt.io/qt-5/stylesheet-examples.html#customizing-qtabwidget-and-qtabbar

carmelopellegrino, 2020-04-10:
actual colouring not implemented, I got it's more clear to work with bold text and white background color

carmelopellegrino, 2020-04-10:
solved in PR #122

closed 2020-04-10 by carmelopellegrino