MechanicalVentilatorMilano/gui #88: investigate memory usage

carmelopellegrino, 2020-04-04:
Check for memory leaks and other memory-related misbehaviours.

Here is a bunch of useful links
- https://medium.com/zendesk-engineering/hunting-for-memory-leaks-in-python-applications-6824d0518774
- https://stackify.com/python-garbage-collection/

marcodeltutto, 2020-04-04:
I have run the `memory_profiler`. Here are the results:
[image]
At time 0 the GUI starts, at the green line I started pushing a lot of buttons in the gui.

Also, I left it running for 1h:
[image]

erykd, 2020-04-22:
This post relates to an observed memory leak on my setup **ONLY** when making a change to the GUI code to "showFullScreen". I understand that on the actual MVM things playout with screen-resolution/window placement, such that Qt's full-screen mode is not needed for the GUI to nicely fill in the available screen. However, if Qt's full-screen mode becomes for some reason needed, please make sure to check for memory leaks in that mode. I, for one, was not expecting to see a memory leak introduced by just placing the GUI in full-screen mode.

My setup:
RPi 2B V1.1
ELECROW 5 inch Capacitive Touch Screen 800x480 TFT LCD Display
```
>> uname -a
Linux raspberrypi 4.19.97-v7+ #1294 SMP Thu Jan 30 13:15:58 GMT 2020 armv7l GNU/Linux
>> cat /etc/debian_version
10.3
>> python3 --version
Python 3.7.3
>>> from PyQt5.Qt import PYQT_VERSION_STR
>>> print("PyQt version:", PYQT_VERSION_STR)
PyQt version: 5.11.3
>>> pyqtgraph.__version__
'0.10.0'
```
Version of code used for test: 2701caa4a06c2b936568a2044b11c4b826310f77

The only line I toggled was:
https://github.com/MechanicalVentilatorMilano/gui/blob/2701caa4a06c2b936568a2044b11c4b826310f77/gui/mvm_gui.py#L63

from "window.show()" to "window.showFullScreen()".

Here is how it looks on my setup:
[image]

Here are the corresponding CPU/memory usage profiles:
[image]