MechanicalVentilatorMilano/gui #92: Support for Canadian communication protocols

bs610, 2020-04-04:
The Canadian version of the MVM uses different hardware components and modified controller code. The communication protocol will be slightly different (a JSON REST API is being proposed instead of a simple get/set), but I do not have an interface control document detailing this yet.

Supporting the Canadian protocol is not a requirement for the Italian MVM.

This feature introduces a new risk that the GUI tries to run with the wrong communication protocol. This risk can be mitigated by keeping the default configuration as the Italian protocol, and requiring a command-line argument to be specified if the Canadian protocol is to be used. The GUI code will already quickly alert the user if unexpected responses are received from the ESP32.

If desired, we could keep the Canadian protocol in a separate branch of the repository, but I don't think that is necessary.

carmelopellegrino, 2020-04-05:
Hi Ben. This Canadian version of the MVM is new to me and it sounds pretty much like a fork (not in the sense of GitHub) rather than an enhancement of the current MVM project.

So let me first understand the difference between MVM and MVM Canada projects, the amount of work required to introduce and maintain the support for two different hardware (which I expect can even diverge at some point), and the role I can have also.  Then we decide whether the best option here is a branch or a fork (in the GitHub sense).

I would really like not to split the GUI project but it seems to happen it is not something under my control.

eto-, 2020-04-05:
I'm not officially aware of any branch in Canada: the GUI will be based on the official version.
You can fork in your own github account if needed. 
This discussion is closed,

closed 2020-04-05 by bs610