MechanicalVentilatorMilano/gui #98: Fix TODO parameters in configuration files

organtin, 2020-04-07:
The following parameters are indicated as TODO in configuration file:

```
    insp_pressure:         'TODO'
    peep_auto:               'TODO'
    support_pressure:  'TODO'
    peep_assist:            'TODO'
    minimal_resp_rate: 'TODO'
```

Fixing these parameters will allow the simulator to reflect settings in the simulated waveforms

carmelopellegrino, 2020-04-07:
Hi Giovanni, you are right. Those are placeholders for the official names that have still to be put in place. I have a list and I was about to file a new issue yesterday on this topic but since one parameter is still missing I didn't submit the issue. I hope this will be fixed soon.

organtin, 2020-04-07:
I just put the issue to track it. I am sure all of you know about that.
This way I can start coding new versions of the simulator as soon as I see
the issue solved.

Thanks

carmelopellegrino, 2020-04-08:
related to issue #99

closed 2020-04-08 by carmelopellegrino