MechanicalVentilatorMilano/gui #151: ESP32 device get calib returns "Valore=..." instead of expected "valore="

bryerton, 2020-04-15:
As a work around  we can for the check_str to lowercase in esp32serial.py

carmelopellegrino, 2020-04-16:
Hi Bryerton, thank you for pointing out. Indeed `get calib` is not a command used in the GUI (did you find it by reading the MVM firmware code?)

Rather than something to be fixed in esp32serial it seems pretty much a bug (typo) of the firmware. I won't make a patch for letting a bug live elsewhere, especially if the patch has a performance penalty.

bryerton, 2020-04-16:
Fair enough, normally I would suggest the upstream (ESP32) fix the bug and
wouldn't have recommended patching here, but as the two are intertwined and
the GUI is easier to patch, I thought I would mention it.

Indeed, I've been going over the ESP32 code, in order to write the tool to
exercise the ESP32 options I had to.

The performance isn't an issue with this patch, this patch has no
significantly measurable performance penalty, as it consists of a single
lower() function called once every 10Hz or whatever your update rate has
been chosen to be for the GUI (max 100 Hz) to call the ESP32.

carmelopellegrino, 2020-04-16:
Actually it is called more often than 10Hz. 10Hz is the rate of the `get_all()` only, then there are more calls (by heart: `get("run")`, `get("mode")`, `get_alarms()`, `get_warnings()`) so the rate increases. And my rpi2 (note: not 4!) has the CPU already running at +90% when reading my Arduino UNO-based mockup. `lower()` means more memory consumption as you duplicate the returned string. That was my overhead concern about.

I would suggest you to open an issue to the official "firmware" repo (https://github.com/NuclearInstruments/MVMFirmware).

closed 2020-04-16 by carmelopellegrino