MechanicalVentilatorMilano/gui #169: Interface freeze and restart after changing patient configuration parameters

josemanuel-cela, 2020-04-20:
Previous steps:
- Raspberry SD card loaded with MvM Raspbian image downloaded from Mega.
- GUI replaced with the last version (merge pull request #163) from github repository.

Steps to reproduce after switching on device:
- Select "new patient" / "start ventilation" 
- Select "menu" / "settings" / "mode settings"
- In "PCV" tab change RR to 20 by clicking on the initial value and using the pop-up table of values.
- Change P_insp to 30.
- Click "apply"

After that GUI is frozen. Few seconds later GUI reboots losing any previous configuration.

carmelopellegrino, 2020-04-20:
Hi Jose Manuel,

couple of questions:
- who made the raspbian image you refer to? The official image, as far as I know, has never put on Mega.
- do you have a complete test bed/bench? I mean, do you have any ESP32 connected to the Raspberry Pi?

josemanuel-cela, 2020-04-20:
Hi Carmelo.

Thank you for your quick response.

We have installed an image that Andrea Abba mentioned in a message he
sent on April 15th to the MVM_ALL list.
He suggested to install the image from this link:

https://mega.nz/#!CUglXaDQ!GdG7wiGVcfgYbDccTzYNWiDsFCl0Ikf2ZuV8xbjf6HQ

This image has an autostart procedure for the GUI.

Besides, I've loaded the last version of the "firmware" into the ESP32,
that was uploaded to the repository yesterday. The ESP is connected to
the Raspberry+Display device installed in the prototype that was sent to
Madrid

But if you recommend me to use the official Raspbian repository, we have
no problem to reproduce the tests.

Best regards.

> Hi Jose Manuel,
> 
> couple of questions:
> 
>     * who made the raspbian image you refer to? The official image, as far as I know, has never put on Mega.
> 
>     * do you have a complete test bed/bench? I mean, do you have any ESP32 connected to the Raspberry Pi? Or use the software mock-up by adding the `fakeESP32` command line argument?

carmelopellegrino, 2020-04-20:
Dear Jose Manuel,

to better understand the situation I need more details. You said that you have updated the firmware on the ESP. Is this version that suggested from Andrea Abba on April 15th?

What is the text output if you run the GUI in a terminal?

I'm almost sure that that version is outdated with respect to the v1.0 tag. I would kindly ask you to wait for the frozen release of the new firmware which will be delivered tomorrow.

josemanuel-cela, 2020-04-21:
Dear Carmelo.

	To complete the information in this issue, the firmware version we are
using is "HW_V3_2020_04_15_00". As far as I know, Andrea didn't
mentioned any specific firmware version rather than recommending to
upgrade it to the lastest one available. But we will wait until a new
firmware release is frozen.

	On the other hand, if I execute the GUI in an external laptop (Debian
Buster) and I try to reproduce the steps mentioned in the first message,
the GUI aborts too. This is the final part of the dump I obtain in the
console:

====================================
ESP32Serial-DEBUG: set alarm 1
Traceback (most recent call last):
  File
"/home/jm/Projects/respirador_mvm/repositorios/MechanicalVentilatorMilano/gui/gui/settings/settings.py",
line 111, in preset_worker
    value = float(value)
ValueError: could not convert string to float: '&20'
KCrash: crashing... crashRecursionCounter = 2
KCrash: Application Name = python3.7 path = /usr/bin pid = 9608
KCrash: Arguments: /usr/bin/python3.7
KCrash: Attempting to start /usr/lib/x86_64-linux-gnu/libexec/drkonqi
from kdeinit
sock_file=/run/user/1000/kdeinit5__0

====================================

	I know that the reason for this could be the differences between
Raspbian libraries and those than are packaged with Debian.

	Thank you for the support.
	Regards.

carmelopellegrino, 2020-04-21:
OK, thank you for the detailed output, now the situation is much more clear to me.

In the above output the line
```
ValueError: could not convert string to float: '&20'
```
indicates that probably there is a nasty character in one of the presets. Could you please check the status of your `gui/default_settings.yaml` file? A `git diff` should be enough to reveal where the problem is.

carmelopellegrino, 2020-04-24:
Any news about this issue? Can we close it?

josemanuel-cela, 2020-04-24:
No news around it, Carmelo.

Please, consider to close it. If we find any new issue we could reopen it.

Thank you.

On 2020-04-24 15:43, Carmelo Pellegrino wrote:
> Any news about this issue? Can we close it?

carmelopellegrino, 2020-04-24:
OK, so for the time being I close it.
Thank you

closed 2020-04-24 by carmelopellegrino