MechanicalVentilatorMilano/gui #184: issues found by pylint

garganti, 2020-04-22:
I run pylint and found many issues.
Some are relates to comments (C:), others propose refactorings (R:).
I propose to fix at least
the errors (E:) - probably related to different versions of pyqt
and the warnings (W:) - this may hide some real bugs

Note that according to the ISO standard we must follow some coding standards adn empoly available tools for sw validation and verification.

For completeness I add the results of pylint

===
 ************* Module communication
C:  1, 0: Missing module docstring (missing-module-docstring)

------------------------------------------------------------------

Your code has been rated at 5.00/10 (previous run: 5.00/10, +0.00)
PyLint: The stdout of the command line is:
 ************* Module communication.esp32alarm
C: 30,47: No space allowed after bracket
        self.alarms = list(filter(lambda x: x, [ self.number & (1<<test) for test in range(32)]))

                                               ^ (bad-whitespace)
C:  1, 0: Missing module docstring (missing-module-docstring)
C: 37, 4: Argument name "n" doesn't conform to snake_case naming style (invalid-name)
R: 48, 8: Unnecessary "else" after "return" (no-else-return)
C: 65,12: Variable name "n" doesn't conform to snake_case naming style (invalid-name)
C: 67,12: Variable name "s" doesn't conform to snake_case naming style (invalid-name)
C: 70,16: Variable name "s" doesn't conform to snake_case naming style (invalid-name)
C: 71,16: Variable name "s" doesn't conform to snake_case naming style (invalid-name)
C: 72,16: Variable name "s" doesn't conform to snake_case naming style (invalid-name)
W: 30, 8: Attribute 'alarms' defined outside __init__ (attribute-defined-outside-init)
C: 80, 0: Missing class docstring (missing-class-docstring)
C:118, 0: Missing class docstring (missing-class-docstring)

------------------------------------------------------------------

Your code has been rated at 6.84/10 (previous run: 6.84/10, +0.00)
PyLint: Executing command line: D:\programmi\python\Scripts\pylint.exe --output-format=text --msg-template='{C}:{line:3d},{column:2d}: {msg} ({symbol})' \mvmgui\gui\mvm_gui.py
PyLint: Executing command line: D:\programmi\python\Scripts\pylint.exe --output-format=text --msg-template='{C}:{line:3d},{column:2d}: {msg} ({symbol})' \mvmgui\gui\numpad\numpad.py
PyLint: The stdout of the command line is:
 ************* Module maindisplay
C:  1, 0: Missing module docstring (missing-module-docstring)
C:  5, 0: Missing class docstring (missing-class-docstring)
I:  5,18: Module 'PyQt5.QtWidgets' has no 'QWidget' member, but source is unavailable. Consider adding this module to extension-pkg-whitelist if you want to perform analysis based on run-time introspection of living objects. (c-extension-no-member)
R:  5, 0: Too few public methods (0/2) (too-few-public-methods)
W:  3, 0: Unused QtGui imported from PyQt5 (unused-import)

------------------------------------------------------------------

Your code has been rated at 3.33/10 (previous run: 3.33/10, +0.00)
PyLint: Executing command line: D:\programmi\python\Scripts\pylint.exe --output-format=text --msg-template='{C}:{line:3d},{column:2d}: {msg} ({symbol})' \mvmgui\gui\presets\presets.py
PyLint: The stdout of the command line is:
 ************* Module alarmsbar
C: 14, 0: Trailing newlines (trailing-newlines)
C:  1, 0: Missing module docstring (missing-module-docstring)
C:  5, 0: Missing class docstring (missing-class-docstring)
I:  5,16: Module 'PyQt5.QtWidgets' has no 'QWidget' member, but source is unavailable. Consider adding this module to extension-pkg-whitelist if you want to perform analysis based on run-time introspection of living objects. (c-extension-no-member)
R:  5, 0: Too few public methods (0/2) (too-few-public-methods)
W:  3, 0: Unused QtGui imported from PyQt5 (unused-import)

------------------------------------------------------------------

Your code has been rated at 1.67/10 (previous run: 1.67/10, +0.00)
PyLint: Executing command line: D:\programmi\python\Scripts\pylint.exe --output-format=text --msg-template='{C}:{line:3d},{column:2d}: {msg} ({symbol})' \mvmgui\gui\settings\gui.py
PyLint: The stdout of the command line is:
 ************* Module communication.threading_utils
C:  4, 0: Multiple imports on one line (traceback, sys) (multiple-imports)
W:  5, 0: Wildcard import PyQt5.QtCore (wildcard-import)
E:  8,20: Undefined variable 'QObject' (undefined-variable)
[... 1623 lines cut ...]
C:409, 4: Missing function or method docstring (missing-function-docstring)
C:412, 4: Missing function or method docstring (missing-function-docstring)
C:418, 4: Missing function or method docstring (missing-function-docstring)
R: 32, 0: Too many public methods (24/20) (too-many-public-methods)
W:  3, 0: Unused QtCore imported from PyQt5 (unused-import)
W:  3, 0: Unused QtGui imported from PyQt5 (unused-import)
W:  6, 0: Unused MainDisplay imported from maindisplay.maindisplay (unused-import)
W: 10, 0: Unused Toolbar imported from toolbar.toolbar (unused-import)
W: 11, 0: Unused Menu imported from menu.menu (unused-import)
W: 12, 0: Unused SettingsBar imported from settings.settingsbar (unused-import)
W: 13, 0: Unused Alarms imported from alarms.alarms (unused-import)
W: 15, 0: Unused AlarmsBar imported from alarms.alarmsbar (unused-import)
W: 16, 0: Unused SpecialBar imported from special.special (unused-import)
W: 18, 0: Unused ToolSettings imported from toolsettings.toolsettings (unused-import)
W: 28, 0: Unused pyqtgraph imported as pg (unused-import)
W: 29, 0: Unused import sys (unused-import)
W: 30, 0: Unused import time (unused-import)
C: 28, 0: third party import "import pyqtgraph as pg" should be placed before "from maindisplay.maindisplay import MainDisplay" (wrong-import-order)
C: 29, 0: standard import "import sys" should be placed before "from PyQt5 import QtWidgets, uic" (wrong-import-order)
C: 30, 0: standard import "import time" should be placed before "from PyQt5 import QtWidgets, uic" (wrong-import-order)
C: 12, 0: Imports from package settings are not grouped (ungrouped-imports)

------------------------------------------------------------------

Your code has been rated at 3.62/10 (previous run: 3.49/10, +0.13)

carmelopellegrino, 2020-04-22:
Ciao Angelo,
do you prefer us to work on reducing the amount of errors and warnings or you are planning to do it in the withing your group?
For me both options are OK, provided that these modifications do not interfere too much with your work. Let us know.

garganti, 2020-04-22:
Hi Carmelo,
we are currently working  on gui/firmaware testing (and CI), so we're not addressing these issues for now.  If someone of you can spend some time on this would be very good. Thanks.

carmelopellegrino, 2020-04-22:
OK, so I'll do the due modifications and release a new tag (say `v1.1`).

carmelopellegrino, 2020-04-26:
Hi Angelo @garganti,
we have produced a new version of the code that incorporates the latest changes and a lot of corrections suggested by `PyLint`. Most of the files (21/29) report a score of 10.00/10, while the worst score among the remaining 8 files is 9.20. Most of the penalties come from `TODO` comments, While others have still to be addressed or can be silenced. We have used `PyLint` 2.4.4 with a custom `.pylintrc` file that can be found under the `gui/` folder and includes `PyQt5` within the whitelisted C-based libraries and silenced some false positives (search for "pylint" throughout the code to list them).

Here is an updated report obtained from the code in the `issue-184` branch:
```

************* Module communication.esp32serial
communication/esp32serial.py:83:4: R0201: Method could be a function (no-self-use)

------------------------------------------------------------------
Your code has been rated at 9.89/10 (previous run: 9.89/10, +0.00)

************* Module special
special/special.py:165:15: W0703: Catching too general exception Exception (broad-except)
special/special.py:48:8: W0201: Attribute '_data_h' defined outside __init__ (attribute-defined-outside-init)
special/special.py:49:8: W0201: Attribute '_config' defined outside __init__ (attribute-defined-outside-init)
special/special.py:50:8: W0201: Attribute '_esp32' defined outside __init__ (attribute-defined-outside-init)
special/special.py:51:8: W0201: Attribute '_messagebar' defined outside __init__ (attribute-defined-outside-init)
special/special.py:87:8: W0201: Attribute '_lung_recruit_timer' defined outside __init__ (attribute-defined-outside-init)

------------------------------------------------------------------
Your code has been rated at 9.20/10 (previous run: 9.20/10, +0.00)

************* Module mainwindow
mainwindow.py:191:2: W0511: TODO: connect to circuit test on ESP (fixme)
mainwindow.py:9:0: E0611: No name 'settings' in module 'settings' (no-name-in-module)
mainwindow.py:10:0: E0611: No name 'settingsfile' in module 'settings' (no-name-in-module)
mainwindow.py:373:15: W0212: Access to a protected member _state of a client class (protected-access)
mainwindow.py:374:16: W0212: Access to a protected member _state of a client class (protected-access)
mainwindow.py:378:16: W0212: Access to a protected member _state of a client class (protected-access)

------------------------------------------------------------------
Your code has been rated at 9.40/10 (previous run: 9.40/10, +0.00)

************* Module frozenplots
frozenplots/frozenplots.py:152:4: C0103: Method name "showEvent" doesn't conform to snake_case naming style (invalid-name)
frozenplots/frozenplots.py:162:4: C0103: Method name "hideEvent" doesn't conform to snake_case naming style (invalid-name)
frozenplots/frozenplots.py:187:8: W0201: Attribute '_cursor' defined outside __init__ (attribute-defined-outside-init)
frozenplots/frozenplots.py:245:8: W0201: Attribute '_cursor' defined outside __init__ (attribute-defined-outside-init)
frozenplots/frozenplots.py:311:8: W0201: Attribute '_cursor' defined outside __init__ (attribute-defined-outside-init)
frozenplots/frozenplots.py:393:8: W0201: Attribute '_cursor' defined outside __init__ (attribute-defined-outside-init)

------------------------------------------------------------------
Your code has been rated at 9.71/10 (previous run: 9.71/10, +0.00)

************* Module start_stop_worker
start_stop_worker.py:117:2: W0511: TODO: Cannot have pop up window (fixme)
start_stop_worker.py:156:2: W0511: TODO: find a good exit point (fixme)
start_stop_worker.py:336:2: W0511: TODO: this should be an alarm (fixme)
start_stop_worker.py:148:4: R0201: Method could be a function (no-self-use)

------------------------------------------------------------------
Your code has been rated at 9.75/10 (previous run: 9.75/10, +0.00)

************* Module alarm_handler
alarm_handler.py:118:4: R0913: Too many arguments (6/5) (too-many-arguments)
alarm_handler.py:174:0: R0902: Too many instance attributes (8/7) (too-many-instance-attributes)

------------------------------------------------------------------
Your code has been rated at 9.84/10 (previous run: 9.84/10, +0.00)

************* Module data_handler
data_handler.py:85:2: W0511: TODO: find a good exit point (fixme)

------------------------------------------------------------------
Your code has been rated at 9.77/10 (previous run: 9.77/10, +0.00)

************* Module settings
settings/settings.py:213:2: W0511: TODO: implement the function to associate to buttons (fixme)
settings/settings.py:12:0: E0402: Attempted relative import beyond top-level package (relative-beyond-top-level)

------------------------------------------------------------------
Your code has been rated at 9.66/10 (previous run: 9.66/10, +0.00)

```

PR #201 (2020-04-27): Issue 184
Fixes #184

that is solving as much as `PyLint` errors as possible.
closed 2020-04-27 by carmelopellegrino