MechanicalVentilatorMilano/gui #216: Misleading dialog in start_stop_worker

bs610, 2020-04-29:
`StartStopWorker._raise_comm_error()` currently creates a dialog with an "OK" button that is tied to `sys.exit()`.

I think two separate dialogs should be produced by this class, depending on where the error came from:
* If `_esp32_io()` was the cause (which is called by a QTimer), the message box should show a Retry button (really a no-op, as the QTimer will cause an automatic retry soon) and an Abort button (which exits the program).
* All the other callers are related to users pressing buttons, so no automatic retry will happen. There should be either an OK button (which does nothing) and an Abort button (which exits).

carmelopellegrino, 2020-04-30:
Communication errors have to be handled differently. That's why there are `TODO` comments all around the place.
Problems are that:
1. popup windows have been asked to be removed (see issue #185)
2. repeated communication errors have the taste of "severe ventilator fault errors" that cannot really be recovered from by just pushing buttons

bs610, 2020-05-03:
Superseded by issue #238 which will create a common interface for handling ESP32 communication errors.

closed 2020-05-03 by bs610