MechanicalVentilatorMilano/gui #217: Misleading dialog created by SnoozeButton

bs610, 2020-04-29:
When snoozing an alarm (in `SnoozeButton._on_click_snooze`), if communication with the ESP32 fails, the user is presented with 2 buttons (Retry and Abort) that both do nothing. 

Abort should be hooked up to `sys.exit`, and either:
* The Retry button should be hooked up to call `self._on_click_snooze` again, or
* The Retry button should be replaced by an Ok button that does nothing

carmelopellegrino, 2020-04-30:
same comment as in [issue #216](https://github.com/MechanicalVentilatorMilano/gui/issues/216#issuecomment-621706454)

bs610, 2020-05-03:
Superseded by issue #238 which will create a common interface for handling ESP32 communication errors.

closed 2020-05-03 by bs610