MechanicalVentilatorMilano/gui #218: Possible failure mode that could cause incorrect settings to be loaded

bs610, 2020-04-29:
There is a failure mode where the doctor could be presented with the "resume ventilation" button, but the most recent settings aren't the ones that are loaded. The sequence is:
* Some settings already exist on disk.
* The doctor makes some changes and starts ventilation, but there was a problem writing the settings to file (in `SettingsFile.store()`). This is silently ignored, and neither the JSON nor MD5 files are updated.
* Something else happens (e.g. power loss or GUI crash), the GUI is restarted, and the doctor clicks "Resume ventilation". The old settings are loaded, rather than the most recent ones.

I suggest that `store()` is changed so that:
* If the settings file does not already exist, and cannot be written to, silently ignore like now
* If the settings file does already exist, but cannot be written to, try to delete the existing settings file and md5 file. If that also fails, alert the user; if the deletion succeeds, either silently ignore, or try to write one more time.

carmelopellegrino, 2020-04-30:
It's not so easy. You can also be prevented to erase the file from disk. Erasing a file is actually a write operation on the FAT.
On the other hand, the probability to have a faulty SD card that failed exactly after having correctly written the md5 file so that the "Resume ventilation" button actually loads something valid, yet incorrect, is quite small, I'd say.

bs610, 2020-04-30:
For your first comment, that's why I said you have to alert the user if you fail to delete the file.

I think you have to consider more than just a disk failure. Unless you can guarantee that no other process can write to the same partition that the settings are stored in, you need to also consider the disk getting full:
* The user starts ventilation. Settings A (and their md5) are written to disk.
* Some time later (whether ventilation is running or not), the disk/partition gets full. The GUI continues to run fine.
* The user makes a change to the settings and starts ventilation. Settings B are not written to disk (as it is full), and the failure is silently ignored. The disk still contains settings A and the md5 for settings A.
* A power cycle or crash happens. The user selects to resume ventilation. Settings A are loaded rather than settings B.

carmelopellegrino, 2020-04-30:
For the moment, no other process writes in that partition. No distribution-specific software uses that partition to store data and for the MVM we foresee only two software, GUI and the Updater. The latter doesn't write to that partition. The GUI always overwrites those two files, i.e. the settings json file and it md5. Both files are (much) less than 1KiB in size and the partition size is 1.3GiB. In case of SD card failures, two are the possible scenarios (based on my personal, yet enough wide, experience):
1. the whole partition gets completely unreadable. In this case, the "Resume ventilation" button is disabled
2. (unusual, but not rare) the whole partition becomes read-only. It's not possible to get the file updated, linux indicates that the file has been written but actually it is all hidden behind the buffers/caches magic. Upon reboot the files are still there exactly as they were before any write/delete operation. In this case there are two possibilities:
2a. the faulty write operation happens during either the write of the settings.json file or during the write of its md5 file. In this case the md5 verification mechanism saves from possibility to load any kind of former configuration, the "Resume ventilation" button is disabled,
2b. the SD stops working exactly after having written the md5 file. In this case it is still possible to load an old, unperfect ventilation parameters set.

The probability of case 2b. is `o(unlikely³)`.

What you propose tries to solve the case in 2b, but there is no real way to be sure the file has been deleted or updated. I have an SD card from a RPi1 B that got damaged by extensively using it with a swap file. It boots, I can modify files, delete them, erase the whole partition, even format it. All operations return successfully. As soon as I reboot the Pi or unplug and replug the SD to the socket it shows the same, perfectly readable files. Deleting before writing is not a valid option.

bs610, 2020-04-30:
Thanks for the extra info. If you're sure that no log files will ever get written to this partition, then you're welcome to close this issue.

carmelopellegrino, 2020-05-01:
Well, I'm fairly sure but as you know all things may change in the future, especially in such a project :)
I've put a `wontfix` label so to mote easily filter in case it will be revived.

closed 2020-05-01 by carmelopellegrino