MechanicalVentilatorMilano/gui #219: Occasional glitches when changing between PCV/PSV modes

bs610, 2020-04-29:
On my machine, when changing between PCV/PSV modes, there are times when the buttons are not repainted, even if the state was correctly changed on the ESP32. This leads to a very dangerous and misleading state. If I leave the menu then come back, the buttons are correct, but that is not very helpful. 

I am not sure why this happens (about 10% of the time for me), but the problem is resolved is `StartStopWorker.toggle_mode()` explicitly repaints the mode and start/stop buttons after changing their text.

carmelopellegrino, 2020-04-30:
This is known to happen - also for other buttons/widgets - on mac OS but not on Linux. Are you using a mac?

bs610, 2020-04-30:
Yes, I'm on a mac. I'll close this issue as you say it's not relevant on linux.

closed 2020-04-30 by bs610