MechanicalVentilatorMilano/gui #221: Trivial static analysis issues

bs610, 2020-04-29:
First, the `.pylintrc` file disables a lot of checks. None of the disabled checks are actually a problem for the GUI code, so the long list after `disable=` should be removed.

Secondly, in `data_filler.py`, one of the pylint exceptions in `add_x_axis_label()` can be removed if `QtWidgets.QGraphicsTextItem()` is used instead of `QtGui.QGraphicsTextItem()`.

Finally, the following trivial issues are currently reported by pylint. Non-trivial issues will be reported separately.

```
alarms/alarmsbar.py:23:0: C0305: Trailing newlines (trailing-newlines)
frozenplots/frozenplots.py:249:8: W0201: Attribute '_cursor' defined outside __init__ (attribute-defined-outside-init)
frozenplots/frozenplots.py:315:8: W0201: Attribute '_cursor' defined outside __init__ (attribute-defined-outside-init)
frozenplots/frozenplots.py:397:8: W0201: Attribute '_cursor' defined outside __init__ (attribute-defined-outside-init)
mainwindow.py:374:15: W0212: Access to a protected member _state of a client class (protected-access)
mainwindow.py:375:16: W0212: Access to a protected member _state of a client class (protected-access)
mainwindow.py:379:16: W0212: Access to a protected member _state of a client class (protected-access)
special/special.py:49:8: W0201: Attribute '_data_h' defined outside __init__ (attribute-defined-outside-init)
special/special.py:50:8: W0201: Attribute '_config' defined outside __init__ (attribute-defined-outside-init)
special/special.py:51:8: W0201: Attribute '_esp32' defined outside __init__ (attribute-defined-outside-init)
special/special.py:52:8: W0201: Attribute '_messagebar' defined outside __init__ (attribute-defined-outside-init)
special/special.py:88:8: W0201: Attribute '_lung_recruit_timer' defined outside __init__ (attribute-defined-outside-init)
special/special.py:10:0: C0411: standard import "import os" should be placed before "from PyQt5 import QtWidgets, uic" (wrong-import-order)
```

carmelopellegrino, 2020-04-30:
First of all, the `.pylintrc` file is the stock one (automatically generated by the `pylint --generate-rcfile` command) plus `PyQt5` has been added to the whitelist to avoid false positives on this C++-based library. Disabled checks have not been touched because they may raise a lot of false positives ant their introduction shall be agreed with the SQA group. @garganti or @a-bombarda, can you comment on this?

garganti, 2020-04-30:
Thanks @bs610. We have to decide which rules, which level and which libraries exclude/include in the Coding Guideline - can you help on that? We risk having many false positives of course. Before further modifying the gui code we want to have a rich test suite for regression testing, and we're currently working on that. We are currently considering other tools, like mypy and bandit.