MechanicalVentilatorMilano/gui #223: Cyclic import

bs610, 2020-04-29:
pylint reports a cyclic import from `communication` -> `communication.esp32serial`. 

Personally I would suggest removing the code from `communication.__init__.py`, and just doing `from communication.esp32serial import ESP32Exception` everywhere instead of `from communication import ESP32Exception`. Half the files already do this anyway. 

Doing this would also solve the only "problem" that pyflakes reports (that an import in `communication.__init__.py` is unused).

This is a very low priority "issue" though, so feel free to close it if you'd prefer to just accept the pylint nagging.

carmelopellegrino, 2020-05-04:
Hi Ben, I have a question: what is the impact to the program of a cyclic import?

bs610, 2020-05-04:
It often indicates a poor code structure. In some cases it can prevent the modules from even being imported. Neither of those apply in this case.