MechanicalVentilatorMilano/gui #238: implement a full-screen system failure message

carmelopellegrino, 2020-05-03:
Here are the relevant situations:
- `system error` (bit 31) is an unrecoverable error indicating that the ventilator is malfunctioning.
- communication error with the ESP32 are severe errors

 -In case of `system error`, the GUI behaviour shall be to:
1. set the **RPi GPIO** pin 17 to `high`. This starts the buzzer+LED alarm system.
2. The area occupied by the `toppane` and `bottombar` widgets shall turn all **solid red**
3. At the center of the `toppane` area a label in bold, white, and big font size shall report `SYSTEM FAILURE\nCall the Maintenance Service`.
4. any further action from the user through the touchscreen must be prevented.

- in case of communication error with ESP32, i.e. an exception of type `ESP32Exception` is raised, the GUI behaviour shall be to:
1. set the **RPi GPIO** pin 17 to `high`. This starts the buzzer+LED alarm system.
2. the area occupied by the `toppane` widget shall turn all **solid red**
3. at the center of the `toppane` area a label in bold, white, and big font size shall report `Severe Communication Error`.
4. in the `bottombar` area a `Retry` button tries to execute again the faulted command (not easy to work on, brainstorming required) for a maximum of 3 times. If the retry succeeds reset the retry counter to `0` and continue, otherwise disable the `Retry` button and be so.

javerom, 2020-05-03:
For point 4, we could implement (at a low level) the saving of the string command on every execution. For example `self.connection.write` in `esp32serial` could be replaced by a wrapper function (perhaps `self.write`) that saves the command to `self._last_cmd` and then executes `self.connection.write`. From this, implementing a "Retry Command" as requested would be trivial.

carmelopellegrino, 2020-05-03:
> For point 4, we could implement (at a low level) the saving of the string command on every execution. For example `self.connection.write` in `esp32serial` could be replaced by a wrapper function [...]

Yeah, problem is that the control flow is anyway interrupted so, as an example, if it is a very nested function asking for one value to be returned it's hard to make it work. I have developed an tremendous idea, though :sunglasses:

closed 2020-05-15 by carmelopellegrino