MechanicalVentilatorMilano/gui #245: Wrong data shown during "system failure"

dd1dd1, 2020-05-04:
I am running "cpp" firmware v0.1 on a bare ESP32 (no MVM control board).
The firmware correctly reports alarm 31 ("system failure").
Per https://github.com/MechanicalVentilatorMilano/mvm-sqa/wiki/FirmwareGuiInterface#alarms
this alarm is raised when firmware cannot read sensors (sensors are absent, in my case).
So all good on the firmware side.

On the GUI side, I see the gui display wrong or erratic values:
- FiO2 is alternating between "-2" and "100"
- Battery is alternating between "0/AC" and "100/Battery"
- Resp rate is "12" (should be "error!")
- Insp/Exp is "1:2" (should be "error!")
- Insp pressure is "15". (should be "error!").

I think when gui detects alarm 31 "system failure" it should ignore all data from the firmware
and display "error!" values.

If the firmware interface is update to indicate which data is valid and which data is invalid,
then you can display valid data.

But cannot do this and alarm 31 should indicate that all data is invalid.

Also it is not clear if "snooze alarm 31" is safe and if it should be permitted.

K.O. olchansk@triumf.ca

dd1dd1, 2020-05-04:
I confirm that erratic FiO2 and battery data displayed by the GUI is caused by erratic data returned by firmware. FiO2 is returned by i2c ADC (absent on my esp32) and bettery data is returned by the supervisor (also absent).

It looks like firmware does not implement an error or alarm flag for read errors from the i2c adc or from the supervisor.

Lacking that, gui code has to protect itself against wrong or erratic data read from the i2c adc and from the supervisor.

In the interface description document these data are labeled "supervisor" and "i2c adc",
but not very clearly, I will try to improve this.
https://github.com/MechanicalVentilatorMilano/mvm-sqa/wiki/FirmwareGuiInterface#get-commands

K.O.

carmelopellegrino, 2020-05-15:
Solved in issue #238

closed 2020-05-15 by carmelopellegrino