MechanicalVentilatorMilano/gui #246: esp32serial.py retries are confused

dd1dd1, 2020-05-04:
The "retry 10 times" code in esp32serial.py is confusing.

This is what it does:
- send command
- read reply, if we do not like the reply, try to read the reply 10 more times.

There is no explanation of why it is done this way, but for the cases I have seen,
it seems to be incorrect:

a) "get all" returns wrong number of values (wrong firmware in esp32): reading the serial line 10 more times is wrong (we already received the reply, there will be no more); sending "get all" command again will not help (we will get the same unexpected result again). Correct action is probably report a global failure "wrong MVM firmware!".

b) "get all" returns garbled data (serial line noise): reading again 10 times will not help. should resend the "get all" command

c) "get unknown command" returns nothing (for many firmware revisions), reading again 10 times will not help, sending command again will not help.

d) "get all" is garbled on the way to esp32 (serial line noise), esp32 receives "!@# $%^", does not reply, again reading 10 time does not help, should resend the command.

K.O. olchansk@triumf.ca

carmelopellegrino, 2020-05-04:
This is going to change and the development is tracked in issue #238

carmelopellegrino, 2020-05-05:
this have to be addressed in issue #238

closed 2020-05-15 by carmelopellegrino