MechanicalVentilatorMilano/gui #247: print "get all" replies

dd1dd1, 2020-05-04:
during code review of the firmware I want to see how firmware replies to different gui commands.

To avoid using a forking the gui repository, would it be possible to add a flag or a config option in esp32serial.py to print the firmware replies to gui commands (a separate flag for "get all" would be nice).

Thanks in advance,
K.O.

carmelopellegrino, 2020-05-04:
Hi Konstantine,

I'm really against adding features (at any level, from trivial ones to complicated) in **production** code for **external debug** purposes (aka things GUI developers don't need :) ). If you need to make a test you can have a patch in your local machine just adding the print line.

I know there is [this project](https://github.com/MechanicalVentilatorMilano/mvm-control) for testing/debug purposes. Maybe you can use that instead of the GUI to read `get all` output.

dd1dd1, 2020-05-04:
Hi, Carmelo, thank you for your reply, I have no problem with you rejecting my request. My concern is with having sufficient debug code in the production version to permit debugging of problems observed during production use. In this specific case, the problem is erratic data displayed by the gui and it is necessary to display the output of "get all" to see if the problem is with gui code or with data from the firmware. We can assume scripts like mvm-control are not available in the production environment, but sensor malfunction or firmware malfunction or gui malfunction is still possible and there should be a way to debug this. That's all. K.O.

carmelopellegrino, 2020-05-05:
Hi Konstantin, I perfectly understand your concerns but consider that:
- during production the operator knows nothing about linux and terminals
- during production there is no keyboard
- we will need to have some kind of "Manufacturer help desk/contact center"
- at the moment, we (as MVM Collaboration) are not eligible as manufacturers :)

So I'd let the real end-manufacturer to define debug/troubleshooting procedures, which can (and should) involve modification of the code provided by the MVM Collaboration.

So, if you don't mind, I'd close this issue. Feel free to reopen it if necessary.
Thanks,
Carmelo

closed 2020-05-05 by carmelopellegrino