MechanicalVentilatorMilano/gui #250: gui crash on unexpected reply to "get alarm"

dd1dd1, 2020-05-04:
Reporting a crash of gui if "get alarm" returns unexpected response:

ESP32Serial-DEBUG: get alarm
Traceback (most recent call last):
  File "/home/pi/gui/gui/alarm_handler.py", line 221, in handle_alarms
    esp32alarm = self._esp32.get_alarms()
  File "/home/pi/gui/gui/communication/esp32serial.py", line 211, in get_alarms
    return ESP32Alarm(int(self.get("alarm")))
ValueError: invalid literal for int() with base 10: 'OK'
Aborted

K.O.

carmelopellegrino, 2020-05-04:
This is also being chased in issue #238, but I have a naive question: how are you testing the gui?

dd1dd1, 2020-05-04:
I am not testing the gui per se. I have an RPi4 with an ESP32 plugged into it, basically it is the same as the MVM machine minus the MVM control board. (I have the TRIUMF MVM control board, but that's a different story). ESP32 is running the MVM firmware (choice of hw_rev3, hw_rev4, V5 and v0.1), the RPi4 is running the GUI. I am not using a touch screen yet, but I will try, eventually, I will know it as I file bugs about touchscreen doing strange things. I have also run the gui against the TRIUMF MVM firmware (from Bryerton) and against the TRIUMF RPi control program (the TRIUMF control board connects directly to the RPi, no ESP32 required)  K.O.

closed 2020-05-15 by carmelopellegrino