MechanicalVentilatorMilano/gui #251: gui does not start if firmware console mode is turned on

dd1dd1, 2020-05-04:
If (for whatever reason) firmware console mode is on ("set console 1"), the gui fails to start, it sits in the infinite loop reading data in ESP32Serial constructor.

It looks like the console mode is incompatible with the GUI (and there will be a proposal to remove it), but one cannot ensure that it is always off by default in every firmware build.

I suggest that gui send the command "set console 0" immediately after opening the serial link (I guess the infinite loop in the constructor has to be moved elsewhere).

As an alternative, this infinite loop should have a timeout and GUI should report an error instead of just sitting there silently.

K.O.

carmelopellegrino, 2020-05-04:
Indeed "set console 1" is incompatible because no one told me that there was such a "feature". Is it really a feature or something that will go away right after commissioning phase?

dd1dd1, 2020-05-04:
Carmelo, it is hard to tell what changes are happening in the firmware, it is best to ask A.Abba directly. All I know is what I see in the code. I know about this "console mode" because early versions of the gui did not work with early versions of the firmware at all because "console mode" was hardwired "always on" (a #define in the code). This is how I know that at needs to be turned off. (and can be turned off). As part of firmware code review, we may ask for the console mode to be removed. K.O.

carmelopellegrino, 2020-05-05:
Hi Konstantin, yes, please. I think that this topic has to be addressed within the firmware review. I'm also against implementing patches to fix bad designs happening elsewhere in the system. Wild-west-style development is not the correct way of pursuing such important initiatives. When the design will be clear, reviewed and accepted we will know how to operate in this issue. Thanks

closed 2020-05-15 by carmelopellegrino