MechanicalVentilatorMilano/gui #252: gui does not report firmware version

dd1dd1, 2020-05-04:
I do not see a button on the gui where the current firmware version is reported. ("get version"). Please add? K.O.

carmelopellegrino, 2020-05-04:
Is it required? By the way, there is a change request board for these things :)

dd1dd1, 2020-05-04:
"is it required?", it better be, how else do you turn the MVM machine off? (But for all I know, the machine does not even have an "on/off" power switch, the MVM sent to TRIUMF certainly does not).

But I do know that if you power-off the RPi (running the GUI), you will corrupt the linux filesystem on the SD card. You have to "shutdown -r now" first.
K.O.

dd1dd1, 2020-05-04:
(I was looking at the wring bug report)

dd1dd1, 2020-05-04:
I think a page reporting the firmware revision, the supervisor revision and the gui revision has to be somewhere. Otherwise, by looking at a machine how can you tell it has the correct software/firmware? K.O.,

carmelopellegrino, 2020-05-05:
Such information is shown in the updater software. It could be useful to have it also shown in the GUI, maybe under the "Maintenance" tab in the settings pane but this change has to be reviewed by the change request board before being implemented.

dd1dd1, 2020-05-07:
Carmelo, how do I submit this change request to the change request board? "gui should report the firmware version and the supervisor version".

carmelopellegrino, 2020-05-07:
Hi Konstantin, you can write an email to Reiner indicating the reason for the requested change. Thanks

closed 2020-05-15 by carmelopellegrino