MechanicalVentilatorMilano/gui #253: gui does not have a shutdown or off button

dd1dd1, 2020-05-04:
I do not see a button on the gui to safely shutdown the MVM machine. it will be added later? if shutdown is by "power-off", the RPi SD card will definitely become corrupted and linux will not boot on next power-on. For now, gui shutdown by ctrl-c seems to work (ctrl-c did not work in some previous versions of the gui, shutdown had to be done by kill -KILL, unsafe when talking to hardware). K.O.

carmelopellegrino, 2020-05-04:
Hi! There is no such button because is a dangerous feature and there is no possibility to turn the whole system off from software. The current way to turn off the ventilator is to cut the power via a protected power switch.

The SD card is partitioned such that the root FS is mounted in read-only and there is a third partition that does synchronous writes to store the previously used settings (those being loaded on "Resume Ventilation").

About the ctrl-c I didn't know. I've always used alt+f4.

dd1dd1, 2020-05-04:
Ok, I will take a second look at how you build the SD card. read-only mounted filesystems should be okey wrt abrupt power-off. For saving settings you can briefly remount read-write, then mount read-only again. In this case, "shutdown" function is not needed. K.O.

closed 2020-05-15 by carmelopellegrino