MechanicalVentilatorMilano/gui #257: Suggestion to change how to handle ESP32 communication errors

a-bombarda, 2020-05-06:
- additional field on the screen "device status"
- remove the "message box" displaying communication errors each time a communication error occurs
This device status field may be green when the device is working properly and does not report errors. After selecting it, the user can view information on the status of each sensor, supply voltage, everything we can check on the device, and report.
Displayed fields may have the appearance: sensor name, next to its sensor status. Each sensor may have an error counter. Such a counter will be increased if e.g. the data from the sensor exceeds the maximum or minimum indication level. If the sensor has a problem with communication with the ESP board or begins to break down and send from time to time, not valid data we will be able to notice this situation.
Each counter should be checked at regular intervals, e.g. 5 min. (parameter in setup or file) also immediately deleted so that it counts again. If the number of errors exceeds the set value (set in the software parameters or configuration file) then the sensor description should change to "error". The sensor description should remain constant until the next check. The device status field can be changed to e.g. yellow. If serious errors occur in red. This is medical equipment, so maybe if any counter has a value greater than zero then the status should light up to yellow. And the status should be In red if more errors appear (counter reach at least 100) within 5 minutes. Similarly, we can check the voltage level of the device's power supply, if it is within the given range, we can write "correct" in the status field "power supply". The "power supply" is the voltage after the regulator which should be constant.

new feature:
- the user does not need to acknowledge each error message
- information on the status of each sensor available online (works, doesn't work)
- quick device diagnostics
- easier servicing

a-bombarda, 2020-05-06:
Error message in response to a communication error
[image]
[image]

Any communication error requires user interaction. With minor communication errors, e.g., the cable between boards that are not properly connected to each other causes that the ventilator is no longer to be used.

a-bombarda, 2020-05-06:
Random data sent as sensor data - both positive and negative numbers. Regardless of whether the sensor data should be in the sensor measurement range
[image]

carmelopellegrino, 2020-05-15:
The gui workflow with respect to errors have been greatly changed in PR #274. Check it out for the new features! 

About the visual suggestions we are not going to restyle the whole thing without a proper change request as it has a huge impact on the development of the gui and on the manual.