MechanicalVentilatorMilano/gui #261: MessageBox class

a-bombarda, 2020-05-06:
The class MessageBox was used to manage alarms received by the ESP32. Now it has been removed, and substituted by red squares at the top of the window.
Despite this, there are other messageboxes used for communication error. Is it correct?

Moreover, the AlarmButtons (which are the red squares at the top of the window) also have s "Warning" mode, which seems to be equivalent to the MessageBoxes that are still present.

The warning AlarmButton is never used. We have to delete that kind of AlarmButton, or substitute the warning MessageBoxes with AlarmButtons.

carmelopellegrino, 2020-05-06:
> The class MessageBox was used to manage alarms received by the ESP32. Now it has been removed, and substituted by red squares at the top of the window.
> Despite this, there are other messageboxes used for communication error. Is it correct?
Correct.
Those are used for sever communication errors (serial link failure) and are about to be removed. See issue #238 for further details.
> Moreover, the AlarmButtons (which are the red squares at the top of the window) also have s "Warning" mode, which seems to be equivalent to the MessageBoxes that are still present.
>
> The warning [...]

To my knowledge (and testing) AlarmButton is used for warnings. Do you have a recent screenshot of this problem happening?

a-bombarda, 2020-05-06:
> Correct.
> Those are used for sever communication errors (serial link failure) and are about to be removed. See issue #238 for further details.

Ok, great! This could be a great improvement also for the testability of the system, since the MessageBox are difficult to be controlled, in terms of contents.

> To my knowledge (and testing) AlarmButton is used for warnings. Do you have a recent screenshot of this problem happening?

I just checked and now the problem seems to be solved. I think that we had a non fully updated version of the GUI on the unit-test branch

carmelopellegrino, 2020-05-06:
OK, good :)

closed 2020-05-15 by carmelopellegrino